7h30th3r0n3/Evil-M5Project #3: Can you help?

loewal, 2024-01-17:
This is what I get trying to compile:

In file included from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/libraries/M5Unified/src/utility/Power_Class.hpp:17,
                 from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/libraries/M5Unified/src/M5Unified.hpp:55,
                 from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/libraries/M5Unified/src/M5Unified.h:5,
                 from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino:36:
/Users/loewal/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/libraries/M5Unified/src/utility/Speaker_Class.hpp:13,
                 from /Users/loewal/Documents/Documenten - iMac van Loe/Arduino/libraries/M5Unified/src/M5Unified.hpp:56:
/Users/loewal/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino: In function 'void setup()':
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino:289:14: error: 'esp_random' was not declared in this scope; did you mean 'srandom'?
  289 |   randomSeed(esp_random());
      |              ^~~~~~~~~~
      |              srandom
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino: In function 'void updateConnectedMACs()':
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino:1480:5: error: 'tcpip_adapter_sta_list_t' was not declared in this scope
 1480 |     tcpip_adapter_sta_list_t adapterList;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino:1482:47: error: 'adapterList' was not declared in this scope
 1482 |     tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |                                               ^~~~~~~~~~~
/Users/loewal/Documents/Documenten - iMac van Loe/Arduino/Evil-M5Core2-v1-1-2/Evil-M5Core2-v1-1-2.ino:1482:5: error: 'tcpip_adapter_get_sta_list' was not declared in this scope
 1482 |     tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "SD.h"
  Used: /Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/SD
  Not used: /Users/loewal/Library/Arduino15/libraries/SD

What is the reason??

loewal, 2024-01-17:
This is what I get:

Serial port /dev/cu.usbmodem569C0158841
Connecting.....
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:51:02:54
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
Failed uploading: uploading error: exit status 2

7h30th3r0n3, 2024-01-17:
Hey @loewal, 
Can you confirm that you setup your M5Core2 with esp32 and M5Stack board and M5unified librairy before flashing the board ?

loewal, 2024-01-18:
Working!
Had to change baudrate to 115200.

7h30th3r0n3, 2024-01-19:
Glad that you find the misconfiguration ! I'm adding it in the installation process to ensure that not happening again ! If it's resolve I'm closing the issue

closed 2024-01-19 by 7h30th3r0n3