7h30th3r0n3/Evil-M5Project #4: Touch screen issue on m5 core s3

cguillaumie, 2024-01-18:
Hello and thank you for your work, the touch screen does not work on the m5 core s3, the program launches well, but it is impossible to navigate in the menus. Can you add the touch screen support?

7h30th3r0n3, 2024-01-19:
Hi ! 
Thx for using the project 😊, 

The problem come from the middle button that is used to select things on Core2, due to the camera on core3 the button as been removed so you can't enter in the menu, I'm actually trying lot of things and few hours to ensure core3 compatibility but I need more time to find a good way to ensure good user experience without asking the user to make a Konami code to press ok button haha, but know that it's planned and pushed as soon I find a good way to do it.

Also know that you can still use the project trough serial command implemented in v1.1.3, in that way you can already test all the features without using any button 😊 command are listed at the end of documentation and you can type help in serial

cguillaumie, 2024-01-19:
Thanks for your feedback, the Konami code brings back old memories! Don't hesitate if you want me to test on core S3

7h30th3r0n3, 2024-01-20:
Hey @cguillaumie, 

I think I find a way to use cores3 properly, let me know if you want to test it, 
It would be maybe better to join the discord to discuss in real time.

cguillaumie, 2024-01-20:
yes, of course I'm volunteer, but I can't reach the discord

7h30th3r0n3, 2024-01-20:
> yes, of course I'm volunteer, but I can't reach the discord

Ho link Is dead .. here it is : https://discord.com/invite/V3FDEC6Q

7h30th3r0n3, 2024-02-08:
closing the issue since it's added on roadmap

closed 2024-02-08 by 7h30th3r0n3