7h30th3r0n3/Evil-M5Project #8: having an issue. its giving me an error

MannyV143, 2024-02-04:
https://github.com/7h30th3r0n3/Evil-M5Core2/blame/f9159044ce2548aac4f7169e7cf65c46a44e804c/Evil-M5Core2-1-1-6.ino#L1509

7h30th3r0n3, 2024-02-04:
Hi ! Thx for using the project !

Can you share to me the model of M5 you are using, the error generated and the exact issue ( won't compile, stuck, loop reboot).

kf5gxz, 2024-02-08:
I'm getting an error as well, screenshot attached. M5FIRE.
[M5FIRE error EVILM5.docx](https://github.com/7h30th3r0n3/Evil-M5Core2/files/14204147/M5FIRE.error.EVILM5.docx)

7h30th3r0n3, 2024-02-08:
> I'm getting an error as well, screenshot attached. M5FIRE. [M5FIRE error EVILM5.docx](https://github.com/7h30th3r0n3/Evil-M5Core2/files/14204147/M5FIRE.error.EVILM5.docx)

For this error you just need to add TinyGpsPlus librairie that you can find on manage librairies,

7h30th3r0n3, 2024-02-21:
closing due to no anwser about the problem

closed 2024-02-21 by 7h30th3r0n3