7h30th3r0n3/Evil-M5Project #10: Portal HTML's

MannyV143, 2024-02-27:
Great work and thank you. Is it possible to get the site in English? They're in French.

I tried logging on to the site provided, but I can't access the site for some reason.

7h30th3r0n3, 2024-02-28:
> Great work and thank you. Is it possible to get the site in English? They're in French.
> 
> I tried logging on to the site provided, but I can't access the site for some reason.

Hey ! Thx for using the project,
yes the phishing pages are French based, but you can modify them very easily to your liking, you just have to change the text in the file with your language of choice, and do it quickly with the upload function, some pages are already in English,

For inaccessible pages, you talk about special pages? or captive portals?

MannyV143, 2024-02-28:
I can't access the link to the url at all for some reason. 
I'm extremely new to all this and I'm just curious on how everything works and I can't figure out for the life of me what is blocking my browser from allowing the page to load.

vassalo, 2024-02-28:
Can you provide some logs from the serial monitor? I think this might help with the troubleshooting.

MannyV143, 2024-03-02:
Lol I'm such an idiot. I fried my laptop last night. It won't power up at all. Went to plug it in and I zapped it. 
Luckily, it was brand new and purchased it just to program/code and learn. So I'll be down for a bit. I'll look for the discord. Thanks for all the help guys. Much appreciated

7h30th3r0n3, 2024-03-02:
> Lol I'm such an idiot. I fried my laptop last night. It won't power up at all. Went to plug it in and I zapped it. Luckily, it was brand new and purchased it just to program/code and learn. So I'll be down for a bit. I'll look for the discord. Thanks for all the help guys. Much appreciated

It happens, take your time, know that it work also on smartphone 😜 I'm closing the issue but feel free to open new one or come on discord server 😉

closed 2024-03-02 by 7h30th3r0n3