7h30th3r0n3/Evil-M5Project #15: Compiler error

loewal, 2024-03-31:
Getting this today:
In file included from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:20,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../common.hpp:22,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:57,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/device.hpp:42,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/M5GFX.h:19,
                 from /Users/loewal/Documents/Arduino/libraries/M5Unified/src/M5Unified.hpp:19,
                 from /Users/loewal/Documents/Arduino/libraries/M5Unified/src/M5Unified.h:5,
                 from /Users/loewal/Documents/Arduino/Evil-M5Core3-1-1-9/Evil-M5Core3-1-1-9.ino:37:
/Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../esp32/../../misc/DataWrapper.hpp:27:10: fatal error: ../../utility/pgmspace.h: No such file or directory
 #include "../../utility/pgmspace.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Can you help?

7h30th3r0n3, 2024-04-01:
Hey ! This error seem to be from M5GFX library, maybe you should try to reinstall M5Stack librairie which contain M5GFX

loewal, 2024-04-02:
Solved it, thanks!

7h30th3r0n3, 2024-04-02:
closing because resolved

closed 2024-04-02 by 7h30th3r0n3