7h30th3r0n3/Evil-M5Project #16: Deauther code

Aro2142, 2024-04-05:
Hello. After a long time of suffering with m5stack, I was able to write code that sends deauth frames and it actually works.
If necessary, I can send you the code for review.
(Written in Arduino IDE)

7h30th3r0n3, 2024-04-07:
Thanks for your work, it's sounds really interesting, feel free to join the discord so we can talk about it, I probably have some questions about modification needed and find a way to integrate it properly in Evil-M5 project 👍

Aro2142, 2024-04-17:
@7h30th3r0n3 
Hello, sorry for the wait, I didn't have time to answer. I described the functions in detail in the code.
I also joined the discord group and will try to answer questions.
I could help with the implementation of this code, but your code is very long, which confuses me.
Now I am busy developing my project, so I cannot provide significant support in your project (((

7h30th3r0n3, 2024-04-17:
@Aro2142 I understand that ! Haha ! The biggest question is about the downgrade of esp32, it's seem that you need to use a special library version? 

Don't worry for the time anybody have a life and project, thx again to pointing this out 🙂

Aro2142, 2024-04-17:
@7h30th3r0n3 I'm not sure, it's just that the Esp32 marauder project used version 2.0.10.  Perhaps you can use a newer version.  I'll try today.  
The libraries are ordinary, you just need to follow the instructions described in the "Marauder" project https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup

Aro2142, 2024-04-18:
Bruh, I can't compile the code for M5Stack Fire, there is a conflict with "int ieee80211_raw_frame_sanity_check", this is bad...

Aro2142, 2024-04-23:
Ok, this is my mistake. The code works even with esp32 version 2.0.11

7h30th3r0n3, 2024-04-28:
Working on my side with latest version of esp32, thanks again for pointing this out, it's was a great help to understand what you need to do for bypassing the protection and send the frame, I need a little bit of time to integrate it but it should be pushed soon 👌🚀

closed 2024-05-04 by Aro2142