7h30th3r0n3/Evil-M5Project #18: M5 Fire keeps restarting

nsbionix, 2024-05-04:
Hey :),
I got myself a M5Fire (the one you also linked) and followed the given installation procedure.
What I did:

Windows 11
Formated SD card 32GB with FAT32.
Installed Arduino IDE
Added the M5 board link in the additional boards manager URLs.
Installed the M5Stack boards and the esp32 boards.
Added the needed libraries.
Patched the firmware with the given script and restartet arduino.
Added the SD-Files to my SD Card and isnerted it in the M5Fire.
I chose the connection port and M5Fire as my board.
Compiled and uploaded without errors.
I get the bootlogo and load into the menu.
If i choose an option in the menu, lets say Bluetooth ON/OFF and open it. The device is loading a sec and then restarting itself.
This problem is with most of the options in the menu.
I can scan Wifi and choose one without restarting but thats kinda it.

Could this be an SD-Card issue? I thought not because i got the bootlogo thats also on the card.

Thanks in advance for your time. Awesome project!
Greetings

7h30th3r0n3, 2024-05-04:
Hey @nsbionix thanks for using the project ! 

Yes it can be SD-card related, to be sure can you try 2 things : 

- try with an sd card lower than 32Go ( 16Go, 8Go) 
- be sure to disable PSRAM in tools menu before uploadind sketch

nsbionix, 2024-05-04:
I actually swapped to a 16GB SD-card and turned the PSRAM off. Seems like PSRAM did the job. Changing the SD-Card only didn't seem to work.
Maybe you cuold add that to the installation instruction also the link to the M5UBoards json.

Thank you very much 👍

7h30th3r0n3, 2024-05-04:
Already added when talked about in last answer,
Thx for the report and glad that work ! 
Enjoy 🥳

closed 2024-05-04 by 7h30th3r0n3