7h30th3r0n3/Evil-M5Project #20: Compiler error Mcore2 1.2.1

loewal, 2024-05-06:
/Users/loewal/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/2.0.16/tools/sdk/esp32/lib/libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; /private/var/folders/hp/fh9kjqk17338tvw9nn2kq9b80000gn/T/arduino/sketches/6AACC484E7667A4B9CDD03FBC46110F5/sketch/Evil-M5Core2-1-2-1.ino.cpp.o:/Users/loewal/Documents/Arduino/Evil-M5Core2-1-2-1/Evil-M5Core2-1-2-1.ino:5044: first defined here
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Can you help?

7h30th3r0n3, 2024-05-06:
Hey ! You probably forget to use the patch script of esp32 firmware as described on the main page which prevent esp32 firmware to send deauth frame, you need to run the script to modify the firmware before compiling, check here : 

https://github.com/7h30th3r0n3/Evil-M5Core2/tree/main/utilities/deauth_prerequisites

loewal, 2024-05-06:
Hi, thanks for the quick answer!
I ran the Mac-script with sudo.
but...
I get the same error.
Ran again, and get the answer that the file is already patched.
What am I missing?

loewal, 2024-05-06:
I even checked the modifications and indeed they were all committed.
But nevertheless that same error...

loewal, 2024-05-06:
> I even checked the modifications and indeed they were all committed. But nevertheless that same error...

and even restarted my Mac Mini

7h30th3r0n3, 2024-05-06:
> > I even checked the modifications and indeed they were all committed. But nevertheless that same error...
> 
> and even restarted my Mac Mini

Seem that the patch not worked, can you try to uninstall and reinstall esp32 firmware, patch it again and retry ? It's sound like it's not applied correctly

loewal, 2024-05-06:
I already did, but will try again.

> > > I even checked the modifications and indeed they were all committed. But nevertheless that same error...
> > 
> > 
> > and even restarted my Mac Mini
> 
> Seem that the patch not worked, can you try to uninstall and reinstall esp32 firmware, patch it again and retry ? It's sound like it's not applied correctly

Still the same. I uninstalled M5stack and reinstalled. Still:
/Users/loewal/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/2.0.16/tools/sdk/esp32/lib/libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; /private/var/folders/hp/fh9kjqk17338tvw9nn2kq9b80000gn/T/arduino/sketches/6AACC484E7667A4B9CDD03FBC46110F5/sketch/Evil-M5Core2-1-2-1.ino.cpp.o:/Users/loewal/Documents/Arduino/Evil-M5Core2-1-2-1/Evil-M5Core2-1-2-1.ino:5044: first defined here
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

loewal, 2024-05-06:
Removed esp32 and M5stack.
installed M5stack which also installed esp32 on demand.
Patched: ok.
Compilation: error!

loewal, 2024-05-06:
Took my old Macbook and installed everything there: same problem!

loewal, 2024-05-06:
I solved it!
But I think there is an 'error' in the instructions:
I found 2 platform.txt:
/Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/2.0.16/
and 
/Users/loewal/Library/Arduino15/packages/m5stack/hardware/esp32/2.1.1/

The patchfile modified the second one...
After manually editing the first one, finally succesful compilation

loewal, 2024-05-06:
I removed the complete m5stack/hardware/esp32/2.1.1/ and compilation was still ok.

Riddle, riddle....

7h30th3r0n3, 2024-05-06:
> I solved it! But I think there is an 'error' in the instructions: I found 2 platform.txt: /Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/2.0.16/ and [...]

Nice ! The patch script are not made by myself, I should take a look to see if anything can cause this issue, maybe the way the search is done omitted the second folder, as far it work, that the principal 😊

closed 2024-05-06 by 7h30th3r0n3