7h30th3r0n3/Evil-M5Project #26: Compilation errors

nash68, 2024-05-27:
Hi,
When I try to compil the code, I have those error messages : 
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void onOffBleSerial()':
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:839:12: error: 'class BluetoothSerial' has no member named 'setPin'
  839 |     ESP_BT.setPin("730303"); // NOT WORKING // WORK ONLY WITH esp v1.0.1 // workaround password in serial
      |            ^~~~~~
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void updateConnectedMACs()':
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2314:3: error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?
 2314 |   tcpip_adapter_sta_list_t adapterList;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
      |   tcpip_adapter_if_t
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2316:45: error: 'adapterList' was not declared in this scope
 2316 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |                                             ^~~~~~~~~~~
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2316:3: error: 'tcpip_adapter_get_sta_list' was not declared in this scope
 2316 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In member function 'virtual void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)':
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:4913:65: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
 4913 |       std::string advData = advertisedDevice.getManufacturerData();
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
exit status 1

Compilation error: 'class BluetoothSerial' has no member named 'setPin'

Any idea ?
Yours.

nash68, 2024-05-27:
It is on a M5Core 
The precompiled version delivered with the M5Burner is working fine, but I woul like to change the Ap name, password, etc...  I will check to remove the line.

nash68, 2024-05-28:
ok. putting comments on the line solves the error (lol).
Any idea for the tcp ones ?

\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void updateConnectedMACs()':
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2314:3: error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?
 2314 |   tcpip_adapter_sta_list_t adapterList;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
      |   tcpip_adapter_if_t
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2316:45: error: 'adapterList' was not declared in this scope
 2316 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |                                             ^~~~~~~~~~~
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:2316:3: error: 'tcpip_adapter_get_sta_list' was not declared in this scope
 2316 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In member function 'virtual void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)':
\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:4913:65: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
 4913 |       std::string advData = advertisedDevice.getManufacturerData();
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?

asafyish, 2024-05-28:
I upgraded esp32 to version 3.0.0 (released recently) and got the same bluetooth errors.
When I downgraded back to 2.0.17 the bluetooth issues were resolved, but then I got 'class m5::Power_Class' has no member named 'Axp192' and 'Serial2' was not declared in this scope.
I think it's related to libraries versions. Maybe publish what versions are you using ?

7h30th3r0n3, 2024-05-28:
> I upgraded esp32 to version 3.0.0 (released recently) and got the same bluetooth errors. When I downgraded back to 2.0.17 the bluetooth issues were resolved, but then I got 'class m5::Power_Class' [...]

Yup that because I use function that are now deprecated in recent version, 
I'm on a fix with most recent version of esp32/m5stack/m5unified librairies

7h30th3r0n3, 2024-05-28:
@nash68 please check version of librairies that you using, v1.2.2 compile fine with :
esp32 version 2.1.1
WiFi version 2.0.0
WebServer version 2.0.0
DNSServer version 2.0.0
SD version 2.0.0 
FS version 2.0.0  
SPI version 2.0.0 
M5Unified version 0.1.14 
M5GFX version 0.1.15
TinyGPSPlus version 1.0.3  
Adafruit_NeoPixel version 1.12.0 
ArduinoJson version 7.0.4 
BLE version 2.0.0  
BluetoothSerial version 2.0.0 

and bypass for deauther

nash68, 2024-05-28:
Hourra !
Removed ALL librairies and put them back one by one.
Install only esp32 board by Espressif (according to this topic : https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup)
Downgrade to 2.0.17 like suggested by  @asafyish 
Apply (manually because the script only search for M5file) bypass for deauther like @7h30th3r0n3 recommanded.

And.... It works ! Thanks a lot guys !

closed 2024-05-28 by nash68