7h30th3r0n3/Evil-M5Project #27: [idea] inspired by M5Stick-NEMO

nash68, 2024-05-28:
Hi, 
thanks for helping me solving the compilation issues I had. Now everything works fine.

May I suggest you to have a look at the M5Stick-NEMO git ?
I made some compairison and found some intersting topic that could maybe improve your Evil-M5Core2 : 

- captive portal are designed to be responsive. On my android phone the templates you provide are very very small.
- the Deauth+Clone (Creates the evil portal with the same SSID and sends deauthentication frames to that AP) is really great !
- on my M5, the Beacon Spam is very very slow...   compared to the WiFi Spam of NEMO

Keep on pushing ! It is a very good job ! I am impressed in the results. Yours.

7h30th3r0n3, 2024-05-29:
Yeah ! It's a well-known and really well made firmware haha ! 
I tested and test everything that M5burner provide and some others project when I got time ! 

About the point that you talking about :

- yup ! Sites provided are only templates and example to explain what you can do with the Evil-M5Project, there are only here to make easier to demonstrate the possibility of doing it, not really to be used in a real attack, but you can add any page you want on sd as you wish, and I made a folder for sites from community to enlarge the number of available web page ☺️

- yeah, actually the portal is only a captive portal that not deauth the initial cloned network, it's planned to add this possibility to ensure a real evil-twin attack 👍 

- Beacon spam was made before I integrate the bypass for sending forged frames on the esp32 so it's a workaround that I found that ensure similar attack without the bypass, but it's less effective than sending forged frames, in other hand it's causing some troubles on things like airodump and others it's build in this way which is not the common way on others project, but it's planned to be refactored since the bypass is now necessary to send deauth.

Glad that you like it ! And yeah it's a good begining, but it's planned to add more ! Until the memory can no more handle it hahaha ! 🚀

nash68, 2024-05-29:
Wait and see then ! 🥇

closed 2024-05-29 by nash68