7h30th3r0n3/Evil-M5Project #30: Cardputer evil 1.2.6 compilation error

loewal, 2024-06-28:
Hi, 
I am getting this error:

> In file included from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:20,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../common.hpp:22,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:57,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/device.hpp:48,
                 from /Users/loewal/Documents/Arduino/libraries/M5GFX/src/M5GFX.h:19,
                 from /Users/loewal/Documents/Arduino/libraries/M5Unified/src/M5Unified.hpp:19,
                 from /Users/loewal/Documents/Arduino/libraries/M5Unified/src/M5Unified.h:5,
                 from /Users/loewal/Documents/Arduino/Evil-Cardputer-v1-2-6/Evil-Cardputer-v1-2-6.ino:38:
/Users/loewal/Documents/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/../esp32/../../misc/DataWrapper.hpp:27:10: fatal error: ../../utility/pgmspace.h: No such file or directory
 #include "../../utility/pgmspace.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Any idea where to repair?

loewal, 2024-06-28:
Will there be a bin for this?

7h30th3r0n3, 2024-06-28:
Mmh sounds like something missing in M5Stack librairie, try to reinstall it maybe, yeah it's should be pushed soon, on the burner and binaries folder

kazz2020, 2024-06-29:
and what about this error?
`c:/users/tomlu/appdata/local/arduino15/packages/m5stack/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\tomlu\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32s3/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; E:\Users\TMP\arduino\sketches\1936C8E96A24E136D181E4FFB7893870\sketch\Evil-Cardputer-v1-2-6.ino.cpp.o:C:\Users\tomlu\Desktop\Evil-M5Core2-main\Evil-Cardputer-v1-2-6/Evil-Cardputer-v1-2-6.ino:5309: first defined here
collect2.exe: error: ld returned 1 exit status`

7h30th3r0n3, 2024-06-29:
You need to apply the deauther bypass that located in utilities folder

kazz2020, 2024-07-01:
yes no problem

closed 2024-07-01 by 7h30th3r0n3