7h30th3r0n3/Evil-M5Project #36: can't fix it

mayday77, 2024-08-08:
C:\Users\********\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32/include/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void snifferCallback(void*, wifi_promiscuous_pkt_type_t)':
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:4508:33: warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
         DynamicJsonDocument json(4096); // Augmenter la taille pour l'analyse
                                 ^
In file included from d:\Evil\libraries\ArduinoJson\src/ArduinoJson.hpp:56,
                 from d:\Evil\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:47:
d:\Evil\libraries\ArduinoJson\src/ArduinoJson/compatibility.hpp:125:58: note: declared here
 class ARDUINOJSON_DEPRECATED("use JsonDocument instead") DynamicJsonDocument
                                                          ^~~~~~~~~~~~~~~~~~~
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void snifferCallbackDeauth(void*, wifi_promiscuous_pkt_type_t)':
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:5039:22: error: variable 'ctrl' set but not used [-Werror=unused-but-set-variable]
   wifi_pkt_rx_ctrl_t ctrl = pkt->rx_ctrl;
                      ^~~~
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void wifi_scan()':
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:5466:13: error: unused variable 'rssi' [-Werror=unused-variable]
     int32_t rssi = WiFi.RSSI(i);
             ^~~~
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino: In function 'void promiscuous_callback(void*, wifi_promiscuous_pkt_type_t)':
D:\Evil\Evil-M5Core2-1-2-2\Evil-M5Core2-1-2-2.ino:5544:22: error: variable 'ctrl' set but not used [-Werror=unused-but-set-variable]
   wifi_pkt_rx_ctrl_t ctrl = pkt->rx_ctrl;
                      ^~~~
cc1plus.exe: some warnings being treated as errors

7h30th3r0n3, 2024-08-08:
Hey ! Can you check the version of esp32 and M5stack libraries, it's seem that there is an update that deprecated some function, try to downgrade it 👍 I'm working on this to be able to flash with most recent version

mayday77, 2024-08-08:
> Hey ! Can you check the version of esp32 and M5stack libraries, it's seem that there is an update that deprecated some function, try to downgrade it 👍 I'm working on this to be able to flash with most recent version

Thank for reply. 

Using board 'm5stack_core2' from platform in folder: C:\Users\******\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1
Using core 'esp32' from platform in folder: C:\Users\***\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1

7h30th3r0n3, 2024-08-10:
> Thank for reply.
>
> Using board 'm5stack_core2' from platform in folder: C:\Users******\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1 Using core 'esp32' from platform in folder: [...]

Hey @mayday77 sorry for the delay, 

I tried to reproduce this issue on my side by never get your result, 
I'm not sure what going on here, 

To be sure here all board an librairies that I use to compile the v1.2.2 :

board 'm5stack_core2' from platform in folder: C:\Users\7h30th3r0n3\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1
core 'esp32' from platform in folder: C:\Users\7h30th3r0n3\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1
WiFi version 2.0.0
WebServer version 2.0.0 
DNSServer version 2.0.0 
SD version 2.0.0
FS version 2.0.0
SPI version 2.0.0
M5Unified version 0.1.14 
M5GFX version 0.1.15 
TinyGPSPlus version 1.0.3 
Adafruit_NeoPixel version 1.12.0 
ArduinoJson version 7.0.4 
BLE version 2.0.0 
BluetoothSerial version 2.0.0

Can you check if there is any divergence that can cause these error please

7h30th3r0n3, 2024-08-19:
@mayday77 any news ?

mayday77, 2024-08-19:
sorry was awk. Will be online for 2-3 days and reproduce all the librarys. Thank you

tobozo, 2024-09-10:
>  cc1plus.exe: some warnings being treated as errors

@mayday77 just set compiler warning level to defaut in arduino prefs, the deprecation warning from ArduinoJSON will still be printed during compilation but it won't stop the compilation when encountered.

[image]

7h30th3r0n3, 2024-09-20:
@mayday77 any news about this ?

7h30th3r0n3, 2024-10-04:
closing no answer

closed 2024-10-04 by 7h30th3r0n3