7h30th3r0n3/Evil-M5Project #47: Error on compiling

Boshimanu, 2024-10-06:
Hi Guys,
Sorry to bother you, but I have a problem, I got this error on compiling:

c:/users/liviu/appdata/local/arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\liviu\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.1.1/tools/sdk/esp32/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\liviu\AppData\Local\Temp\arduino\sketches\6920A2DCDFD247A0A3712B0598E60663\sketch\Evil-M5Core2-1-2-2.ino.cpp.o:C:\Users\liviu\Desktop\Evil-M5Core2-main\Evil-M5Core2-1-2-2/Evil-M5Core2-1-2-2.ino:5086: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Compilation error: exit status 1

If I delete this function from the code, it compile correctly:
//extern "C" int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3) {
//  if (arg == 31337)
//    return 1;
//  else
//    return 0;
//}

 I have an M5Stack core2.
I've tried it on my mac and pc, and I get the same error.
Thank You!

Boshimanu, 2024-10-07:
Update: so I've used uploaded the firmware to the device and I think it's working properly, I do not get any errors, and everything seams to work great. The question is: what this function does (ieee80211_raw_frame_sanity_check) and if it's ok to use it without.
Thanks again.

7h30th3r0n3, 2024-10-07:
ieee80211_raw_frame_sanity_check function is a function from esp32 firmware that prevent to send dangerous forged frame, 
So you deleted the redifinition function in the code that send true for any frames to bypass this restrictions,
it's used in deauther and most of forged frames like beacon or probes, 

so yeah it should seem to work, but you should see no effect when deauther is used because it cant send this kind of frames, 

You probably dont apply the deauther bypass that reconfigure esp32 firmware to be able to redifine this problematic function, 

more information and script for automation are provided here : 
https://github.com/7h30th3r0n3/Evil-M5Core2/tree/main/utilities/deauth_prerequisites

Boshimanu, 2024-10-07:
Thank you so much for your response, I did not much testing after installing, so you're definitely right. 
Your response should send me in the right direction.🙏
I saw that there are .bin compiled files already, is it ok to use that? I think the file is: Evil-M5Core2-1-2-2.bin.
Thank you again, cause I finally found a purpose for my M5Core2.🙏
Thank You Sir!