7h30th3r0n3/Evil-M5Project #59: Network hijack/DHCP starvation issues

Jethro-San, 2024-12-16:
whenever I try to use network hijacking or DHCP starvation after it detects a server it will just restart evil. I have tried to fix this, and the compiling was fine with no issues. I am using the cardputer

7h30th3r0n3, 2024-12-16:
Hey ! 

it's sound like a crash from something not anticipated, 
someone already reported the same issue i'm currently investigating, it's seem to happen when the DHCP answer a bad ip address,
it should be fixed soon

7h30th3r0n3, 2025-01-01:
Hey @Jethro-San, last version as been updated the DHCP starvation, can your try again to see if you still have the issue ? I was unable to reproduce the crash on my side but I changed few things that can be your problem here

ringerbart, 2025-03-14:
Here I have the same issue. After the DHCP server is found, It reboots. The DHCP server on my test network was on ip 192.168.10.253  I changed DHCP server to 192.168.1.1 but still get the same reboot so it's not an ip range issue.

7h30th3r0n3, 2025-04-14:
> Here I have the same issue. After the DHCP server is found, It reboots. The DHCP server on my test network was on ip 192.168.10.253 I changed DHCP server to 192.168.1.1 but still get the same reboot so it's not an ip range issue.

Some modification have been done in V1.4.0 can you please check again ?

ringerbart, 2025-04-15:
No more reboots here but I don't get a NAK on my testrouter. It keeps going. Maybe the router isn't vulnerable?  It's a noname chinese mini router.

[image]

[image]

> Some modification have been done in V1.4.0 can you please check again ?

7h30th3r0n3, 2025-04-16:
> No more reboots here but I don't get a NAK on my testrouter. It keeps going. Maybe the router isn't vulnerable? It's a noname chinese mini router.

Well it depend of the DHCP itself, If your network is a /24 then it's probably not vulnerable, because there is more than 254 IP ACK and it make a rotation for new IP forgetting the old one,
if it's a /16 then it take 65536 IP address so it's not NAK yet, 

Also not all DHCP send NAK when starvated, so you can stay pressed on enter to go to the next step to see if the DHCP is starvated

7h30th3r0n3, 2025-05-16:
closing for fixed