7h30th3r0n3/Evil-M5Project #60: Compiler error on CYD beta ino

loewal, 2025-01-04:
Hi,
First of all, I installed XP2046_Bitbang_Slim, because XP2046_bitbang generated several errors.
But now there are different errors:

/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In function 'int getConnectedPeopleCount()':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:916:3: error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?
  916 |   tcpip_adapter_sta_list_t adapterList;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
      |   tcpip_adapter_if_t
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:918:45: error: 'adapterList' was not declared in this scope
  918 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |                                             ^~~~~~~~~~~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:918:3: error: 'tcpip_adapter_get_sta_list' was not declared in this scope
  918 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In function 'void updateConnectedMACs()':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:2992:3: error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?
 2992 |   tcpip_adapter_sta_list_t adapterList;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
      |   tcpip_adapter_if_t
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:2994:45: error: 'adapterList' was not declared in this scope
 2994 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |                                             ^~~~~~~~~~~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:2994:3: error: 'tcpip_adapter_get_sta_list' was not declared in this scope
 2994 |   tcpip_adapter_get_sta_list(&stationList, &adapterList);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In function 'void sniffMaster()':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:7416:32: error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]
 7416 |   if (esp_now_register_recv_cb(OnDataRecvSniffer) != ESP_OK) {
      |                                ^~~~~~~~~~~~~~~~~
      |                                |
      |                                void (*)(const uint8_t*, const uint8_t*, int) {aka void (*)(const unsigned char*, const unsigned char*, int)}
In file included from /Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:78:
/Users/loewal/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-083aad99-v2/esp32/include/esp_wifi/include/esp_now.h:163:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
  163 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
      |                                    ~~~~~~~~~~~~~~~~~~^~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In member function 'virtual void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:8214:65: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
 8214 |       std::string advData = advertisedDevice.getManufacturerData();
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In member function 'virtual void SkimmerAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:8642:73: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
 8642 |       std::string deviceAddress = advertisedDevice.getAddress().toString();
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:8643:56: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
 8643 |       std::string deviceName = advertisedDevice.getName();
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~^~
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino: In function 'void startWardivingMaster()':
/Users/loewal/Documents/Arduino/Evil-CYD-Beta/Evil-CYD-Beta.ino:9496:28: error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]
 9496 |   esp_now_register_recv_cb(OnDataRecv);
      |                            ^~~~~~~~~~
      |                            |
      |                            void (*)(const uint8_t*, const uint8_t*, int) {aka void (*)(const unsigned char*, const unsigned char*, int)}
/Users/loewal/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-083aad99-v2/esp32/include/esp_wifi/include/esp_now.h:163:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
  163 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
      |                                    ~~~~~~~~~~~~~~~~~~^~
Multiple libraries were found for "SD.h"
  Used: /Users/loewal/Library/Arduino15/packages/esp32/hardware/esp32/3.1.0/libraries/SD
  Not used: /Users/loewal/Documents/Arduino/libraries/SD
exit status 1

Compilation error: 'tcpip_adapter_sta_list_t' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?

Can you help?

Thanks!!

7h30th3r0n3, 2025-01-04:
Hi @loewal, thanks for using the beta project ! 
Can you try to downgrade the esp32 librairie below 3.x, 
M5stack use esp32 v2.1.1 maybe it's the issue here

loewal, 2025-01-05:
Hi, you talk about M5stack...but I'm trying to get CYD working...

7h30th3r0n3, 2025-01-05:
yes i'm talking about m5stack esp32 version.... 
because it's from the main project ported for CYD and originaly for M5stack that have there own version of esp32,
if you check  https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json the version of esp32 is m5stack-2.1.3 so the lastest version of esp32 is 2.1.3 for m5stack, due to this if you use the normal esp32 firmware that is at 3.1.0, and dont have anymore  tcpip_adapter_sta_list_t you experiment some issue, 

so yeah downgrading the esp32 board version to below 3.x should do the job and make you able to compile.
i'm not talking about the librairie but about the esp32 board itself that also have a version which in your case seem to be above 3.x

loewal, 2025-01-06:
I am very confused now..how can I downgrade my CYD (ESP32-2432S028R)?
Thnx!

Aidenwebb, 2025-02-02:
> yes i'm talking about m5stack esp32 version.... because it's from the main project ported for CYD and originaly for M5stack that have there own version of esp32, if you check [...]

Adding more specifics to this as a rookie to Arduino / CYD / Arduino IDE

Boards Manager installed packages:

- esp32 v2.0.17 (esp32 v2.1.1 is not available in my Arduino IDE)

[image]

Targetting `ESP32 Dev Module`

Library Manager installed libraries

- SD v1.3.0
- Adafruit Neopixel v1.12.4
- Arduinojson v7.3.0
- ESPping v1.0.5
- IniFile v1.3.0
- LibSSH-ESP32 v5.3.1
- TFT_eSPI v2.5.43
- Tjpg_Decoder v1.1.0
- TinyGPSPlus v1.0.3
- XPT2046_Bitbang v1.0.2

[image]

Getting error `Compilation error: no matching function for call to 'XPT2046_Bitbang::XPT2046_Bitbang(int, int, int, int, int)'`

I'll update as I troubleshoot

Talismancer, 2025-02-04:
I adjusted the function call (line 49?) to reflect the function:

// Initialize the touch screen using software SPI
XPT2046_Bitbang ts((uint8_t)XPT2046_MOSI, (uint8_t)XPT2046_MISO, (uint8_t)XPT2046_CLK, (uint8_t)XPT2046_CS); 

The call doesn't need the XPT2046_IRQ parameter. I also cat all the parameters to uint8_t but probably not required.