7h30th3r0n3/Evil-M5Project #61: Unit GPS v1.1 no gps data.

tomi671, 2025-01-13:
My cardputer cannot receive any GPS data from the V1.1 GPS module. The older GPS module has communication at 96000 baud.  This v1.1 needs 115200 baud.  On another firmware, after switching the speed to 115200, the gps works.  It would be great to add a speed selection option in the firmware.

7h30th3r0n3, 2025-01-13:
Hi ! Yeah i'm aware of this issue and it's already implemented on the next version that should be pushed soon 😊

tomi671, 2025-01-14:
Hi and thank you, it's great.  Because the old version of the gps module is not available here.

7h30th3r0n3, 2025-01-20:
> Hi and thank you, it's great. Because the old version of the gps module is not available here.

the new version of Evil-Cardputer contain the selection of baud rates in settings, I'm closing the issue now it's implemented

closed 2025-01-20 by 7h30th3r0n3