7h30th3r0n3/Evil-M5Project #65: Evil Portal not showing the "Sign In" alert

Blazz3, 2025-02-04:
I using a Cardputer. When I deploy the Evil Portal and connect my Samsung s24+ to the portal, I get a "no internet connection" message. A sign-in alert should appear, as with other firmware like Nemo.

Examples of firmware that effectively shows a "sign-in" alert when the Captive Portal is enabled:
https://github.com/SailAway33/M5_captive/blob/main/CaptPort/CaptPort.ino
https://github.com/n0xa/m5stick-nemo/blob/main/m5stick-nemo.ino

I'm not sure if the issue is related to the "WiFi.mode(WIFI_MODE_APSTA)" as the other firmware use normally WIFI_MODE_AP, additionally I think for Android devices a valid response for the /generate_204 page is needed.

7h30th3r0n3, 2025-02-05:
Hi @Blazz3 ! 

Thx for the report, `WiFi.mode(WIFI_MODE_APSTA)` should be the same as `WiFi.mode(WIFI_MODE_AP)` but with STA enable in parrallele, it should work like the others project even if i'm using it, 

Can you provide the version of the cardputer you using ? 

There is a bug on <1.3.9 when the AP name/password are in config file but the AP is not near, it cause instabilities from webserver and DNS, but fixed on v1.3.9. 
It work on my android 15 and any device tested on my side, so it should work on your side too

Blazz3, 2025-02-06:
Hi @7h30th3r0n3 thank you for your prompt response!
I'm using the latest 1.3.9 version (Cardputer). I even compiled the code myself to test some changes in the code (from WIFI_MODE_APSTA to WIFI_MODE_AP, adding the "/generate_204" page, etc.) but no success. I don't know if the issue is only for Samsung Phones.

ringerbart, 2025-03-14:
I have the same problem on a samsung phone. I have the core2 and cardputer and problem is the same on both. On a xiaomi phone it works.  Strangly enough, on nemo, the portal does work on my samsung.

sanchetkore, 2025-04-06:
For this to work on samsung the ip and gateway shoukd be 172.0.0.1 . 192.168.4.1 wont work on samsung.

7h30th3r0n3, 2025-04-06:
Interesting 🤔 I'm gonna do some test and probably implement a way to change local network configuration

7h30th3r0n3, 2025-04-14:
> [@sanchetkore](https://github.com/sanchetkore) is right, I replaced the default 192.168... to 172.0.... and the popup alert from the evil portal is working on a S23+

Ok thanks for the feedback, im gonna make a function to change IP adress of provided network

7h30th3r0n3, 2025-09-13:
> > > [@sanchetkore](https://github.com/sanchetkore) is right, I replaced the default 192.168... to 172.0.... and the popup alert from the evil portal is working on a S23+
> > 
> > 
> > Ok thanks for the feedback, im gonna make a function to change IP adress of provided network
> > [@7h30th3r0n3](https://github.com/7h30th3r0n3)
> > This hasn't been implemented yet correct? If so I'm not seeing it when running the current release from the Launcher repository. If I'm not mistaken the Evil Portal on other devices lets you choose between the two IPs (192.168.4.1 or 172.0.0.1) when you launch it, but if that is too hard/long to implement then maybe just a config file line could serve in the interim? It would be great to get this working because I use this as a practical demo to teach friends/families just how easy it can be to get tricked and why you should not use free wifi. I can use Bruce to do this but it would be awesome to have it work in this, I like the tools you put together in this.

Not it's not implemented for now because there is more than just changing the IP address, a lot of things are hardcoced to 192.168.4.1 but I probably found a solution, it's probably gonna be pushed on v1.4.6 which should arrive soon

closed 2025-09-13 by 7h30th3r0n3