7h30th3r0n3/Evil-M5Project #66: Cardputer: Devices can't connect to evil portal AP

slg328, 2025-02-05:
Devices refuse to connect to the Captive Portal AP unless the SSID is set to default "Evil-Cardputer" Stops working when I rename SSID to something other than default.

7h30th3r0n3, 2025-02-05:
tested on my side and it work, by settings the ssid with set ssid option or change it trough probes list, 
What version are you using ?

Blazz3, 2025-02-06:
From my side is working too (Carputer 1.3.9).

slg328, 2025-02-06:
1.3.9

slg328, 2025-02-06:
It seems to only fail when I'm in crowded places with many connections around.

7h30th3r0n3, 2025-02-14:
Tested on a place with more than 50+ AP near me, and it seem to work, have you tested again ? is it something that happen regulary ?

slg328, 2025-02-14:
The AP was named and it hasn’t worked every time ive tried. When the name
wasn’t edited and it showed up as Evil-Cardputer it worked perfectly

7h30th3r0n3, 2025-02-14:
Ok to be sure an able to reproduce, you doing: 
- set wifi SSID ( with one you want)
- start captive portal 
And it don't show up when there is multiple AP around you ? Or you never seen any custom SSID ?