7h30th3r0n3/Evil-M5Project #71: Add support for ESPBOY

coppaxamsukk, 2025-03-12:
Espboy is a diy esp8266 based project that's similar to cardputer but available from long ago. 

Can you port the awesome project for espboy?
https://www.espboy.com/

[image]

7h30th3r0n3, 2025-03-18:
Sounds like a really cool project, but I dont see any pentest/cybersecurity functionnalities, is there anything is your mind that can be ported on Evil-m5Project ?

coppaxamsukk, 2025-03-18:
> Sounds like a really cool project, but I dont see any pentest/cybersecurity functionnalities, is there anything is your mind that can be ported on Evil-m5Project ?

I'm sure evil m5 will work, 
(And esp32 based esp32boy is on development https://community.espboy.com/t/esp32boy-the-next-advanced-espboy/300/9 )some features already included in espboy.

 like GSM phone
GPS navigator
Meteostation
LORA messenger
RFID/NFC reader/repeater
Sub1Ghz inspector
Sub1Ghz scanner
MOD/S3M/XM/IT player
AY38910/12 Yamaha chip player
Web radio
FM radio
Walkie-talkie
Meccano rover
Flower watering
Ice fishing automation
Volt/Amper meter
CW radio beacon
RSS reader
Digital ruler
Pinger
Digital thermometer
Heart rate meter
WiFi analizer
WiFi sniffer
WiFi packet monitor
WiFi pinger
WiFi deauther
IR blaster
Geiger counter
I2C scanner
MP3 player

closed 2025-04-14 by 7h30th3r0n3