7h30th3r0n3/Evil-M5Project #80: Functionality issue

goodfellasthai, 2025-04-20:
I love this firmware and exactly what I was looking for in all ESP32 products, I have noticed one minor issue with it,

When using 'Handshake/Deauth Sniffing' leaving in Static mode is usually fine but this only scans one particular channel.

When I try to use m to go to Mode: Auto, then the cardputer switches itself off after a couple of minutes switching through channels, I have thought of taking a look at the source code but I am sure someone on here would know the answer faster than me, I would just like Handshake/Deauth to continue running on all channels collecting handshakes.  Is this possible?

P

7h30th3r0n3, 2025-04-21:
Hi 👋 

Yeah this is an old bug that I can't figure out, 
The screen just shutoff but program continues to run in the back, it's just the screen, I'm gonna double check it to see if there is any chance to get the problem 😊

goodfellasthai, 2025-04-21:
Hugely appreciate your response and is brilliant that it continues to gather the handshakes in the back ground.
Like I said I think this is mt favourite of all ESP projects I have looked at, I am probably not up to the level of coding you guys are yet, but in the future as I increase my knowledge would love to become involved in the the project.   As a general IT consultant I see so many uses for this device and this code line.

Kind Regards

Paul

goodfellasthai, 2025-04-22:
Strangest of things if I leave the USB plugged in the cardputer while switched on in static mode it does not switch off, and I think it is physically switching off when it times out in static mode without the USB C plugged in - possibly some sort of device power timeout while in static mode.
I am going to fork the source and have a look myself I genuinely love this project.
Are you guys working on anything else interesting like this you would recommend me having a look at?
Kind Regards
Paul

7h30th3r0n3, 2025-05-11:
Hey Paul ! 

I think I found the bug ! It's seem to happen when you use multiple access to screen in a callback which can cause the SPI to crash, I need to change a lot of thing in that part of the code to fix the problem, but it should be fixed soon 👍

I recommend that you use wifi raw sniffer instead of this function until it's fixed even it's not automatic about switching channel it's probably better to prevent screen crash