7h30th3r0n3/Evil-M5Project #84: M5StackFire V2.7 - Can't boot "RTC not found" on v.1.3.9

InWeirdPeril, 2025-05-10:
I didn't find this one in the issues list yet and I just ran into it.

After flashing the Firmware Core2-v1.3.9 or Fire-v1.3.9 to my M5StackFire, it will no longer boot but instead stop with the message "RTC not found".

I tested flashing Core2-v1.2.2, which works fine.

I ran into these issues with both using M5 Burner directly as well as using the M5Launcher.

7h30th3r0n3, 2025-05-11:
Due to RTC introduction, which is not available on fire, it's now two separate firmware on M5burner you should find it on Evil-Fire/Core instead of using Core2 one

InWeirdPeril, 2025-05-11:
Ah, I figured it may something like that. 

I really am not complaining, just trying to understand :)

I found the one on M5Burner. Though after having a look into the .ino files for Core2, I'd like to compile my own for the Core/fire to change some hardwired (Recommended ones at that) things, but there does not seem to be a specifc .ino for the Core/Fire. I assume it's compiled from the Core2? I found case switches depending on the board in the code.

And just to clarify for my own sanity, 

the project .bin "[Evil-Fire-v1.3.9.bin](https://github.com/7h30th3r0n3/Evil-M5Project/blob/main/binaries/Evil-Fire-v1.3.9.bin)"
is effectively mislabeled as it also seem to require an RTC despite it being for the Fire?

I am just trying to sort things out for myself to better understand it.

1.) repository supplied .bin for Fire is wrong as it requires a RTC
2.) Only M5Burner .bin works, albeit a... weird version (1.7) that's actually older?
2.) No specific .ino for the Fire