7h30th3r0n3/Evil-M5Project #85: Missing "sites" folder causes M5StackFire to lock up and force restart

InWeirdPeril, 2025-05-10:
Just something I did notice if you haven't copied the sd card files or try to use a different folder (Because I tried to use M5Launcher and organize my files a bit on the SD card. If it's missing, the devices locks up and force restarts. 

Very little impact on anything really, just wanted to mention it.
Selecting a folder or file would be nice in the future, but hardly important :)

7h30th3r0n3, 2025-05-11:
Yeah there is no SPIFFS on Evil-M5project which mean SDcard and files are a requirement for working properly, there is multiple bug on many function that can crash, freeze or get outside of the function,  also all folder path are hardcoded which mean that you need to have it in a very specific place, I'm aware of the problem caused by the use of multiple firmwares, it's planned to make a one folder on sd card to prevent any conflict with others firmware when you use launcher

InWeirdPeril, 2025-05-11:
Alright, thank you very much for the information.

closed 2025-05-11 by InWeirdPeril